arduino-libraries/Servo #3: Update Servo library for more standard default pulse widths

agdl, 2016-07-12:
_From @bhuang10000 on December 23, 2014 19:33_

The Servo library uses a default pulse width of 544 uS to 2400 uS -- with a zero/neutral position at 1472 uS? 

This does not coincide with any standard servos that I've seen. Most servos (like HiTech or other generic ones) operate on a rage from 900 uS to 2100 uS with a neutral position (90 deg) at 1500 uS. The "default pulse width" is set to 1500, but this is not the same as 90 degrees on the servo.

Suggest changing the default settings to align with 900 uS and 2100 uS --

#define MIN_PULSE_WIDTH       900     // the shortest pulse sent to a servo  
#define MAX_PULSE_WIDTH      2100     // the longest pulse sent to a servo 
#define DEFAULT_PULSE_WIDTH  1500     // default pulse width when servo is attached

_Copied from original issue: arduino/Arduino#2509_

dok-net, 2020-01-21:
It could be obvious from the namings, MAX_xxx on the hand sind, DEFAULT_xxx on the other, that there is an issue with outer boundaries of the SW implementation and what should be safe (as in don't catch fire) defaults min and max.
In my case, I have had a brand manufacturer servo (Tower Pro SG90) destroyed by the defaults timings used here.

So, please fix the library to allow whatever MIN/MAX is deemed to be in use "out there", but set the DEFAULTs for the 0° and 180° timings in µs to either the specification's 1000µs/2000µs, 900µs/2100µs from this issue's author, or my 800µs/2190µs empirically determined on 5 servos.

dok-net, 2020-01-21:
About 1500µs and 90°, it would seem to be a general agreement that manufacturers share, despite their use of varying timings for 0° and 180°.

dok-net, 2020-01-21:
Dear @cmaglie, I am kindly asking you to revisit this issue and read what happened to me (on ESP8266). This issue obviously exists across all platforms. For standard timing servos, 1000/1500/2000µs is the specified timing, for highspeed servos, the center at DEFAULT_PULSE_WIDTH 1500 is completely different (1/2, 1/4), so I can reasonably say that the current defaults, somewhat confusingly called MIN/MAX, are not a great service to anyone.
The change that I am suggesting would be to substitute:
```
#define MIN_PULSE_WIDTH       544     // the shortest pulse sent to a servo  
#define MAX_PULSE_WIDTH      2400     // the longest pulse sent to a servo 
```
by
```
#define DEFAULT_MIN_PULSE_WIDTH       1000     // the shortest pulse sent to a servo  
#define DEFAULT_MAX_PULSE_WIDTH      2000     // the longest pulse sent to a servo 
```
across the library.
If you consider this appropriate (please do so :-) ), I'd make available a PR to that effect.
Otherwise, I am also fine with the values 900/2100 suggested by this issues original author, whereas I maintain my personal measurements gave 800µs/2190µs for a full sweep that is safe, in that it does not force any of the servos I tried outside their mechanical range of operation.

rileyjshaw, 2020-09-23:
> Dear @cmaglie, I am kindly asking you to revisit this issue and read what happened to me (on ESP8266). This issue obviously exists across all platforms. For standard timing servos, 1000/1500/2000µs is the specified timing, for highspeed servos, the center at DEFAULT_PULSE_WIDTH 1500 is completely different (1/2, 1/4), so I can reasonably say that the current defaults, somewhat confusingly called MIN/MAX, are not a great service to anyone.
> The change that I am suggesting would be to substitute:
> 
> ```
> #define MIN_PULSE_WIDTH       544     // the shortest pulse sent to a servo  
> #define MAX_PULSE_WIDTH      2400     // the longest pulse sent to a servo 
> ```
> 
> by
> 
> ```
> #define DEFAULT_MIN_PULSE_WIDTH       1000     // the shortest pulse sent to a servo  
> #define DEFAULT_MAX_PULSE_WIDTH      2000     // the longest pulse sent to a servo 
> ```
> 
> across the library.
> If you consider this appropriate (please do so :-) ), I'd make available a PR to that effect.
> Otherwise, I am also fine with the values 900/2100 suggested by this issues original author, whereas I maintain my personal measurements gave 800µs/2190µs for a full sweep that is safe, in that it does not force any of the servos I tried outside their mechanical range of operation (0° - 90° - 180°)

@dok-net just to add a cross-reference, there's already a PR open with your proposed changes https://github.com/arduino-libraries/Servo/pull/24. Sadly there's been no response on the PR over the past 1.5 years, but I'm keeping my fingers crossed :)

d-a-v, 2020-10-27:
@rileyjshaw just in case you have not noticed, your proposed changes have conflicts.

rileyjshaw, 2020-10-27:
@d-a-v that's because it's been sitting unreviewed for 1.5 years. As I mention in the PR comments I'm happy to fix the conflicts, but I'd like some input from a maintainer before spending more time on it.

vdneut, 2021-12-13:
Fully support the above. Additionally add a feature to set DEFAULT_PULSE_WIDTH. Currently it is set to 1500