arduino-libraries/Servo #12: Servo compilation error in platform io

adaror, 2018-03-23:
hey,
i'm using platform io IDE, and nodemcu.
and when im trying to compile the library i get this error message:
"This library only supports boards with an AVR, SAM, SAMD, NRF52 or STM32F4 processor."

there is something to do to fix it? it is very important.

thanks.

adaror, 2018-03-23:
my .ini file is:
[env:nodemcuv2]
platform:espressif8266
board: nodemcuv2
framework = arduino

PaulStoffregen, 2018-03-23:
This is soooo the wrong place to report this issue.  ESP8266 isn't an Arduino product, **and** you not even using Arduino's IDE.

Both of those projects are on github.  You should contact them.  No point wasting the time of developers who don't support your hardware and didn't write the software you're using!

adaror, 2018-03-25:
OK first of all i don't know who do you think you are talking like that, there are better ways to tell a men that he his wrong, specially when i'm new with all of this stuff so i don't know who you are or how good are you but apparently by reading your stupid answer i guess you don't know anything yourself because a developer with a little dignity would share links of the right places to post this problem. i don't want even to hear what you have to say and you welcome to delete your stupid answer.
i there are anyone normal in this forum?

jeaoks, 2018-07-29:
Hi adaror,

i have the same issue, and i am also newbie. I have ESP32 with OLED display board.

I have found out that servo.h library does not work for some reason.

But i found anohter servo library which works, but the examples are imho a bit sluggish. I am trying to set this up:
https://randomnerdtutorials.com/esp32-servo-motor-web-server-arduino-ide/

but the problem is that this particular example uses servo.h library. And in this example they are using ESP32 and that same library...

PaulStoffregen, 2018-07-29:
This is Arduino's issue tracker.  You're reporting a problem with a library someone else wrote (or modified), when used with boards and products Arduino doesn't make or sell.  Can you understand how this is so the wrong place to complain?

If you're using the library from that tutorial, open your issue on it's issue tracker!
https://github.com/RoboticsBrno/ESP32-Arduino-Servo-Library/issues

closed 2019-05-07 by per1234