arduino-libraries/Servo #15: Does not work with NRF52 device

shwetankshrey, 2018-08-23:


shwetankshrey, 2018-08-23:
Hi

I am using @sandeepmistry 's Arduino Core for an nRF52 device. When I try to compile the Sweep example, it fails with the error message : "This library only supports boards with an AVR, SAM, SAMD, NRF52 or STM32F4 processor". Upon digging further into Servo.h, I find the following conditions : 

> #if defined(ARDUINO_ARCH_AVR)
> #include "avr/ServoTimers.h"
> #elif defined(ARDUINO_ARCH_SAM)
> #include "sam/ServoTimers.h"
> #elif defined(ARDUINO_ARCH_SAMD)
> #include "samd/ServoTimers.h"
> #elif defined(ARDUINO_ARCH_STM32F4)
> #include "stm32f4/ServoTimers.h"
> #elif defined(ARDUINO_ARCH_NRF52)
> #include "nrf52/ServoTimers.h"
> #else
> #error "This library only supports boards with an AVR, SAM, SAMD, NRF52 or STM32F4 processor."
> #endif

It is possible that the ARDUINO_ARCH_NRF52 is not defined in @sandeepmistry 's core. How can I use the Servo library for my nRF52 device?

per1234, 2018-08-23:
`ARDUINO_ARCH_NRF52` is correct for [Arduino nRF52 Boards](https://github.com/arduino-org/arduino-core-nrf52) (the core for the Primo). The equivalent macro for [sandeepmistry/arduino-nRF5](https://github.com/sandeepmistry/arduino-nRF5) is `ARDUINO_ARCH_NRF5`.