arduino-libraries/Servo #23: Disable homing of servo on attach (or slow down movement)

nospam2k, 2019-04-01:
Is there a way to disable homing of server when servo.attach()? In my use, it whip around so fast that pulls the device over on when the ESP8266 is powered on.

Either that or a way to slow down the movement so it doesn't move so fast.

dadul96, 2019-05-15:
As far as I know, there is no technical way to achieve this desired behavior with any servo library or code. The problem is that the library doesn't receive a position/angle signal from the servo. There is not even a position sensor connected to the microcontroller. Everything is controlled/regulated by the servo itself.

So there are three possible solutions:
* If you have a programmable servo, you could try to activate the soft start feature.
* If you start the servo always from the same position, you could clone this library and change the `DEFAULT_PULSE_WIDTH` macro in "Servo.h" from **1500** to your **desired angle** (converted to microseconds).
* You do the same as in solution 2, but instead of setting your desired angle directly, you set it to **0**. After that, you would have to manually increment from the (known) start position to your desired position. This also allows you to determine the speed of this movement.

nospam2k, 2019-05-15:
I'll look into some combination of 2 and 3 as that seems like what I'm looking for. What I'm not understanding is there must be some feedback from the servo because on servo.attach() the servo moves to some starting point. From your description, that starting point has something to do with 1500. A slower increment to this 1500 seems to be what I'm after. The servo is an SG90 if that makes any difference.

PaulStoffregen, 2019-05-15:
This question is probably more appropriate for the forum, since it's really a question about how to use Servo.

Since the communication is unidirectional, and Arduino sends only the desired position, there's nothing that can be done inside the Servo library.  However, you can better design your program, and that is the sort of topic meant for the forum, not github issues.

closed 2019-05-15 by nospam2k