arduino-libraries/Servo #61: servo not working on nano 33 ble sense

1198159, 2020-07-31:
i cant compile my code, even if there is nothing in it and the library is just included. 
I get this message:

C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\libraries\Servo\src\mbed\Servo.cpp:7:11: error: 'DigitalOut' in namespace 'mbed' does not name a type
     mbed::DigitalOut   *pin;
           ^~~~~~~~~~
C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\libraries\Servo\src\mbed\Servo.cpp:8:11: error: 'Timeout' in namespace 'mbed' does not name a type
     mbed::Timeout      timeout;  // calls a callback once when a timeout expires
           ^~~~~~~
C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\libraries\Servo\src\mbed\Servo.cpp:9:11: error: 'Ticker' in namespace 'mbed' does not name a type
     mbed::Ticker       ticker;   // calls a callback repeatedly with a timeout
           ^~~~~~
C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\libraries\Servo\src\mbed\Servo.cpp: In constructor 'ServoImpl::ServoImpl(PinName)':
C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\libraries\Servo\src\mbed\Servo.cpp:13:7: error: 'pin' was not declared in this scope
       pin = new mbed::DigitalOut(_pin);
       ^~~
C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\libraries\Servo\src\mbed\Servo.cpp:13:7: note: suggested alternative: 'min'
       pin = new mbed::DigitalOut(_pin);
       ^~~
       min
C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\libraries\Servo\src\mbed\Servo.cpp:13:17: error: expected type-specifier
       pin = new mbed::DigitalOut(_pin);
                 ^~~~
C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\libraries\Servo\src\mbed\Servo.cpp: In destructor 'ServoImpl::~ServoImpl()':
C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\libraries\Servo\src\mbed\Servo.cpp:17:9: error: 'ticker' was not declared in this scope
         ticker.detach();
         ^~~~~~
C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\libraries\Servo\src\mbed\Servo.cpp:17:9: note: suggested alternative: 'time'
         ticker.detach();
         ^~~~~~
         time
C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\libraries\Servo\src\mbed\Servo.cpp:18:9: error: 'timeout' was not declared in this scope
         timeout.detach();
         ^~~~~~~
C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\libraries\Servo\src\mbed\Servo.cpp:18:9: note: suggested alternative: 'timer_t'
         timeout.detach();
         ^~~~~~~
         timer_t
C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\libraries\Servo\src\mbed\Servo.cpp:19:16: error: 'pin' was not declared in this scope
         delete pin;
                ^~~
C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\libraries\Servo\src\mbed\Servo.cpp:19:16: note: suggested alternative: 'min'
         delete pin;
                ^~~
                min
C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\libraries\Servo\src\mbed\Servo.cpp: In member function 'void ServoImpl::start(uint32_t)':
C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\libraries\Servo\src\mbed\Servo.cpp:24:7: error: 'ticker' was not declared in this scope
       ticker.attach(mbed::callback(this, &ServoImpl::call), 0.02f);
       ^~~~~~
C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\libraries\Servo\src\mbed\Servo.cpp:24:7: note: suggested alternative: 'time'
       ticker.attach(mbed::callback(this, &ServoImpl::call), 0.02f);
       ^~~~~~
       time
C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\libraries\Servo\src\mbed\Servo.cpp: In member function 'void ServoImpl::call()':
C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\libraries\Servo\src\mbed\Servo.cpp:28:9: error: 'timeout' was not declared in this scope
         timeout.attach(mbed::callback(this, &ServoImpl::toggle), duration / 1e6);
         ^~~~~~~
C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\libraries\Servo\src\mbed\Servo.cpp:28:9: note: suggested alternative: 'timer_t'
         timeout.attach(mbed::callback(this, &ServoImpl::toggle), duration / 1e6);
         ^~~~~~~
         timer_t
C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\libraries\Servo\src\mbed\Servo.cpp: In member function 'void ServoImpl::toggle()':
C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\libraries\Servo\src\mbed\Servo.cpp:33:8: error: 'pin' was not declared in this scope
       *pin = !*pin;
        ^~~
C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\libraries\Servo\src\mbed\Servo.cpp:33:8: note: suggested alternative: 'min'
       *pin = !*pin;
        ^~~
        min
In file included from C:\Users\Alex Stedman\Documents\ArduinoData\packages\arduino\hardware\mbed\1.1.6\variants\ARDUINO_NANO33BLE/pins_arduino.h:3:0,
                 from C:\Users\Alex Stedman\Documents\ArduinoData\packages\arduino\hardware\mbed\1.1.6\cores\arduino/Arduino.h:86,
                 from C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\libraries\Servo\src\mbed\Servo.cpp:3:
C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\libraries\Servo\src\mbed\Servo.cpp: In member function 'uint8_t Servo::attach(int, int, int)':
C:\Users\Alex Stedman\Documents\ArduinoData\packages\arduino\hardware\mbed\1.1.6\cores\arduino/macros.h:28:40: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
 #define digitalPinToPinName(P)      (P >= PINS_COUNT ? NC : g_APinDescription[P].name)
C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\libraries\Servo\src\mbed\Servo.cpp:64:44: note: in expansion of macro 'digitalPinToPinName'
   servos[this->servoIndex] = new ServoImpl(digitalPinToPinName(pin));
                                            ^~~~~~~~~~~~~~~~~~~
exit status 1
Error compiling for board Arduino Nano 33 BLE.

Whats the issue?

1198159, 2020-08-01:
nvm i fixed it by downloading the latest from github, rather than the arduino manager

per1234, 2020-09-04:
Thanks for the report  @1198159. As you discovered, this bug was already fixed by https://github.com/arduino-libraries/Servo/pull/60.

I have now made a 1.1.7 release of the library with the fix, so within the next several hours, the version you can download from Library Manager will work with the Nano 33 BLE.

closed 2020-09-04 by per1234