arduino-libraries/Servo #70: enlarge range for angle from 0 to 359

dsyleixa, 2021-01-24:
hello,
will it be possible to enlarge the range for angle e.g. from 0 to 359 or at least from 0 to 270?
The default servo max angle= 180 should be possible to be changed arbitrarily for that purpose.

hugoinfante83, 2021-03-24:
I think is the same issue that i have...

int Servo::read() // return the value as degrees
{
  return map(readMicroseconds()+1, SERVO_MIN(), SERVO_MAX(), 0, 180);
}

uhertlein, 2021-11-21:
@dsyleixa, @hugoinfante83 You are aware that this won't actually change the travel range of the servo, right?
It would merely make it easier to match your code to your hardware - i.e. if you have a servo with a 270 degree travel range and then you can use that value in your code.

IMHO the mentioning of `angle` and `in degrees` in this library is a bit unfortunate, as `value` represents neither in the `write` or `read` methods.

Picking 180 for the maximum position can lead people to assume that it is in degrees, but 1000, 255, or 100 (like percent, but poorer resolution) might have been better choices.

dsyleixa, 2021-11-21:
IMO the servo position to write or read should always match to its actual angle in degrees and nothing else, for either servo with either travel range.

uhertlein, 2021-11-21:
That would be ideal, and that's why I was asking about your expectations/use-case.

Unfortunately, the library has no way to find out what the physical travel range of an attached servo is.
So the programmer would need to know the travel range of the servo and could then configure that in the `Servo` class.

That would be an easy change (e.g. adding the max 'angle' to the constructor) and could also be done in a backwards-compatible manner by using 180 as the default.

dsyleixa, 2021-11-22:
yes, I know the travel range (e.g., 180 or 270 or 360), the lib is just to handle the angles as htey are programmed. 
e.g., 180 is supposed to be 180° at either servo, and 270 then 270° at servos with 270 oreven  larger ranges.
Of course an inappropriate programmed value would cause malfunctions, but that's up to the programmer and the builder.

uhertlein, 2021-11-24:
I'd suggest adding an additional `attach` override like so:

`uint8_t Servo::attach(int pin, int min_value, int min, int max_value, int max);`

- `min_value` is the corresponding value when the servo is at the `min` pulse length
- `max_value` is the corresponding value when the servo is at the `max` pulse length

The default `attach` call would then call
`this->attach(pin, 0, MIN_PULSE_WIDTH, 180, MAX_PULSE_WIDTH);`
for backwards compatibility.

The programmer could adjust this to fit the problem, like having a range from 0..270 or even -45..45.

dsyleixa, 2021-11-24:
IMO that's a bad idea, the lib should provide the correct handsome features out of the box for either servo range.

uhertlein, 2021-11-24:
What sort of an API do you have in mind?

You could always provide convenience functions on top of that, like `Servo::attach(int pin, RangeType rangeType)` that calls the `attach` override with [0,90], [0,180], [0,270], or [0, 360].

But for my use-case having [-45, 45] would be ideal and IMO there's no reason that the API should prohibit that.