arduino-libraries/Servo #71: feature request: Servo class: servo.write(int angle, float rotationspeed)

dsyleixa, 2021-01-24:
feature request: Servo class: servo.write(int angle, float rotationspeed)
angle: degrees (endpoint)
rotationspeed: degrees per second
e.g.,
```
servo.write(90); // let the servo move to position 90° immediately
delay(1000);
servo.write(180, 10.0); // moves the servo from the current position (90°) to position 180° by a rotation speed of 10°/sec

```
The function must not be blocking, so that a following command can be started immediately before the further movement  has been finished (simultaneous movement)

servo1.write(180, 10.0);
servo2.write(45, 5.0);

macMikey, 2024-05-04:
Dumb question: I don't think we can know the position of a normal servo.
If all we can do is command the servo to move to a position, how would we control the speed?

dclause, 2024-05-04:
The speed is usually controlled by manually setting the change in position at a given pace. 
For instance if you require positions 1°, 2°, 3°,...90° every say 100ms, you would then do a 90° move at a speed of 9sec/90°. By opposition of requiring 90° directly, the move being done at an unknown speed, depending on the servo capacity, speed, force, the power applied couple to move, etc...

macMikey, 2024-05-04:
i understand what is being requested, and i love this idea. it's unclear to me how it can be done. at least at startup.
the normal servo does not have a read pin, so the adruino cannot know the servo's position, or the rate at which it is moving (or the rate at which it can move). adurino can command a move to a position, but that's it.
`servo.read()` only returns the last position that the servo was commanded to move to, not where it is (because the hobby servo can't share that information). at startup, for instance, that value would be undefined.
once a homing or first move has achieved, though, things are much simpler.