arduino-libraries/Servo #74: warning for mbed_nano architecture

montoyamoraga, 2021-04-09:
i have been using this library with an Arduino Nano 33 BLE Sense,
and Arduino released a new core for it, where the architecture is named mbed_nano
so now i have a warning that says
"
WARNING: library Servo claims to run on avr, megaavr, sam, samd, nrf52, stm32f4, mbed architecture(s) and may be incompatible with your current board which runs on mbed_nano architecture(s).
"

withinboredom, 2021-05-04:
@montoyamoraga just curious, how did you get this to compile? I receive the following error:

> error: 'DigitalOut' in namespace 'mbed' does not name a type

withinboredom, 2021-05-04:
I guess it helps to install the library.

montoyamoraga, 2021-07-31:
this was already resolved

closed 2021-07-31 by montoyamoraga

bilal697, 2021-09-10:
> this was already resolved

How to solve for mbed_nano-architecture.

per1234, 2021-09-10:
@bilal697 as mentioned above it was already resolved quite some time ago. Just make sure you are using the latest 1.1.8 version of the Servo library. You can check the version and update the library in the Arduino IDE's Library Manager (**Sketch > Include Libraries > Manage Libraries...**).

If you have any questions, you're welcome to post on the Arduino Forum:
https://forum.arduino.cc/