arduino-libraries/Servo #83: Conflict with virtualwire.h library

saurabhlanje199, 2021-08-22:
It shows 
```
\Servo.cpp.o (symbol from plugin): In function `ServoCount':
(.text+0x0): multiple definition of `__vector_11'

\VirtualWire\VirtualWire.cpp.o (symbol from plugin): first defined here
collect2.exe: error: ld returned 1 exit status
Multiple libraries were found for "Servo.h"
```
The problem is that both `Servo.cpp` and `VirtualWire.cpp` try to use the same interrupt vector. Arduino Uno has only a single 16-bit timer, and Servo.cpp library seems to want it. The problem is that VirtualWire library also wants to use the only 16-bit timer available.

PassiSchoppi, 2022-04-20:
I experience the same issue with the "Adafruit_VL53L0X.h" library.
Have you found a quick fix?

fxf06, 2022-06-11:
There is an answer [here](https://electronics.stackexchange.com/questions/67967/problems-using-servo-h-and-virtualwire-h-in-arduino).
Unfortunately, the two libs use the same timer. Looking for a solution (some mentioned ServoTimer2, but does not this to work either, and I don't how to convert angles into ms...). Seems to be a recurring issue :-(