arduino-libraries/Servo #91: Make output frequency of the PWM signal adjustable.

NAmmann, 2022-01-17:
It happens, that some servos or other motor controllers require the PWM signal to be outputted at a specific frequency. These frequency my be between 5 and 400 Hz. 
It would be awesome to configure that output frequency on initialization of a single servo or on a global scale.

silvan2468, 2022-04-20:
You can do this by change the refresh rate in servo.h.
But it is magic for me, why the possible frequeny decreases as more servo objects are generated. With only on servo, 333 Hz ar possible. With 2 servos 200 Hz and with 4 servos about 150 Hz. Why is this?
I would understand this with 8 bit / 16 MHz prozessors. But why also with the much faster 32 bit processors with frequencies >48 MHz? 
Is this library up to date or an old dinosaurus? ;)

hpssjellis, 2023-02-13:
Looks similar to 

https://github.com/arduino/arduino-examples/issues/62

Can the line below be accessed from the sketch to be changed before the Servo include file is loaded? 

Note a refresh interval of 20000 us is 0.02 seconds is 1/0.02 = 50 Hz. 
So 333 HZ needs a refresh interval of 3003 us  (1/333 = 0.003003 seconds)

```
#define REFRESH_INTERVAL    20000     // minimum time to refresh servos in microseconds 
```

razvanphp, 2024-09-17:
It seems that someone did it: https://github.com/arduino-libraries/Servo/issues/123#issuecomment-1935327466

But for me, it does not change no matter what value I put there; I suspect editing `Servo.h` has no impact, as a pre-compiled Servo.a lib is being actually embedded in the sketch.

This might also be hinted by the fact that Arduino IDE shows this file as read-only:

[image]