arduino-libraries/Servo #112: Outdated `renesas` architecture ID used in library metadata

per1234, 2023-06-27:
The `architectures` field of [the `library.properties` metadata file](https://arduino.github.io/arduino-cli/dev/library-specification/#library-metadata) is a list of identifiers of platform architectures the library is compatible with.

A single repository is used to host the codebase of the platform used by the Renesas microcontroller-based UNO R4 and Portenta C33 boards. During the initial development of the platform, it was distributed as a single `arduino:renesas` platform, meaning the architecture ID was `renesas`. When support for the boards of that architecture was added to the library, this architecture ID was added to the library metadata.

Since that time, the distribution approach has changed to providing a separate platform for each of the board series, each with its own architecture ID:

| Platform | Architecture ID |
|-|-|
|**Arduino UNO R4 Boards**| `renesas_uno`|
|**Arduino Renesas Portenta Boards**| `renesas_portenta`|

🐛 The library metadata was not updated to use the new architecture IDs. This causes a warning to be shown in the verbose compilation output when the library is compiled for one of the Renesas microcontroller-based boards:

```text
WARNING: library Servo claims to run on avr, megaavr, sam, samd, nrf52, stm32f4, mbed, mbed_nano, mbed_portenta, mbed_rp2040, renesas architecture(s) and may be incompatible with your current board which runs on renesas_uno architecture(s).
```

🐛 This architecture ID mismatch can also impact dependency resolution because the match between the `architectures` field and the architecture ID of the board being compiled for [is used as a factor by the sketch build system](https://arduino.github.io/arduino-cli/dev/sketch-build-process/#architecture-matching).

### To reproduce

1. Select one of Arduino's Renesas microcontroller-based boards that are compatible with the Servo library from the **Tools > Board** menu in Arduino IDE.
1. Select **File > Preferences** from the Arduino IDE menus (**Arduino IDE > Settings** for macOS users).
1. Check the box next to "**Show verbose output during: ☐ compilation**" in the "**Preferences**" dialog.
1. Click the **<kbd>OK</kbd>** button.
1. Compile the following sketch:
   ```cpp
   #include <Servo.h>
   void setup() {}
   void loop() {}
   ```
1. Wait for the compilation to finish.
1. Examine the contents of the "Output" panel at the bottom of the Arduino IDE window.

🐛 A spurious warning is shown, something like:

```text
WARNING: library Servo claims to run on avr, megaavr, sam, samd, nrf52, stm32f4, mbed, mbed_nano, mbed_portenta, mbed_rp2040, renesas architecture(s) and may be incompatible with your current board which runs on renesas_uno architecture(s).
```

### Expected behavior

Library metadata contains the architecture IDs of all compatible platforms.

### Library version

1.2.0

per1234, 2023-06-28:
Closing as fixed by https://github.com/arduino-libraries/Servo/commit/4def8a2d8fb1fd06503a312bdddb55bc0368f8db

closed 2023-06-28 by per1234

MrYsLab, 2023-06-28:
Tested and indeed resolved. Thanks for the quick fix.