arduino-libraries/Servo #115: Change Timer1 to Timer3 on ATmega32u4-based dev boards

CoolBots, 2023-07-13:
As of version 1.2.1 of the library, ATmega32u4-based boards are defined to use `Timer1` raher than `Timer3`. This configuration conflicts with `millis()` and `micros()` functions. While not too difficult to change, it does make the library more difficult to use in educational settings (we're a robotics/programming school for kids). If possible, could `Timer1` be changed to `Timer3` for these devices? The specific device we are using is an [Adafruit ItsyBitsy 5V](https://www.adafruit.com/product/3677). 

It would be no issue to submit a PR, of course, provided there isn't a strong reason to keep it as `Timer1`. Likewise, if there is a workaround that is fairly straightforward and does not require modification of the library that we are not aware of, please let us know. 

It is worthwhile mentioning that we have seen other examples of this choice of `Timer1` instead of `Timer3` being an issue on 32u4 boards, most notably from _Pololu Robotics and Electronics_: https://www.pololu.com/docs/0J57/8.b

_CodeRef_: https://github.com/arduino-libraries/Servo/blob/master/src/avr/ServoTimers.h#L41-L43

per1234, 2023-07-13:
Hi @CoolBots. Thanks for your report.

>As of version 1.2.1 of the library

This statement seems misleading to me. No change to the timer used by ATmega32U4-based boards was introduced in the 1.2.1 release. In fact, the last relevant change was made back in 2012:
https://github.com/arduino/Arduino/pull/102

and released in [Arduino IDE 1.0.2](https://github.com/arduino/Arduino/releases/tag/1.0.2) (at that time the Servo library was distributed as a part of the IDE installation), and in [the 1.0.0 release of the standalone Servo library](https://github.com/arduino-libraries/Servo/releases/tag/1.0.0) after the library was split out from the IDE project.

CoolBots, 2023-07-14:
My apologies, I merely meant to tie the  issue to the version of the library I discovered it in. You're right though, it's misleading, as it sounds like a new issue in this version, which is not the case. I'll be more careful with my wording; nevertheless, the issue exists - what are your thoughts on my proposed change to use `Timer3` instead of `Timer1`?