arduino-libraries/Servo #123: maybe the servo library not compatible with Arduino H7 yet, how can i fix this issues?

ghiathkamel, 2024-02-05:
Hello :slight_smile:

i got this errors when verification the code, it looks the servo library not compatible with Arduino H7 yet, how can i fix this issues:
\Arduino\libraries\Servo\src\mbed\Servo.cpp: In member function 'void ServoImpl::start(uint32_t)':
\Arduino\libraries\Servo\src\mbed\Servo.cpp:31:66: warning: 'void mbed::TickerBase::attach(F&&, float) [with F = mbed::Callback<void()>]' is deprecated: Pass a chrono duration, not a float second count. For example use `10ms` rather than `0.01f`. [since mbed-os-6.0.0] [-Wdeprecated-declarations]
       ticker.attach(mbed::callback(this, &ServoImpl::call), 0.02f);
mbed_portenta\4.0.10\cores\arduino/mbed/drivers/include/drivers/Ticker.h:92:10: note: declared here
     void attach(F &&func, float t)
          ^~~~~~
Arduino\libraries\Servo\src\mbed\Servo.cpp: In member function 'void ServoImpl::call()':
Arduino\libraries\Servo\src\mbed\Servo.cpp:35:80: warning: 'void mbed::TickerBase::attach(F&&, float) [with F = mbed::Callback<void()>]' is deprecated: Pass a chrono duration, not a float second count. For example use `10ms` rather than `0.01f`. [since mbed-os-6.0.0] [-Wdeprecated-declarations]
         timeout.attach(mbed::callback(this, &ServoImpl::toggle), duration / 1e6);
                                                                                ^
In file included from packages\arduino\hardware\mbed_portenta\4.0.10\cores\arduino/mbed/mbed.h:84:0,
                packages\arduino\hardware\mbed_portenta\4.0.10\cores\arduino/mbed.h:8,
                Arduino\libraries\Servo\src\mbed\Servo.cpp:5:
packages\arduino\hardware\mbed_portenta\4.0.10\cores\arduino/mbed/drivers/include/drivers/Ticker.h:92:10: note: declared here
     void attach(F &&func, float t)

ghiathkamel, 2024-02-05:
the GPT4 told me to do that: To fix your warning, you need to replace the float parameter with a chrono duration in your code. For example, instead of writing:

timeout.attach(mbed::callback(this, &ServoImpl::toggle), duration / 1e6);

You should write:

timeout.attach(mbed::callback(this, &ServoImpl::toggle), duration * 1us);

This will make your code more precise and compatible with future versions of Mbed OS. but after editing the servo.cpp i got new errors, i dont know what to do?

per1234, 2024-02-05:
Hi @ghiathkamel. Thanks for submitting this issue.

>I got these errors when compiling the code

It is important to understand the difference between warnings and errors. A warning is the compiler telling you there is something in the code that could possibly cause a problem but doesn't cause the compilation to fail. An error is a problem with the code that causes compilation to fail. In this case you have posted warnings, not errors. In order to avoid confusion, we should be careful to use the appropriate terminology when communicating about technical subjects.

The code in the library that causes the warnings should be fixed, and we will use this issue to track that task. However, as long as the library is functional you should not let the warnings distract you from progressing with your project.

ghiathkamel, 2024-02-06:
> Hi @ghiathkamel. Thanks for submitting this issue.
> 
> It is important to understand the difference between warnings and errors. A warning is the compiler telling you there is something in the code that could possibly cause a problem but doesn't cause the compilation to fail. An error is a problem with the code that causes compilation to fail. In this case you have posted warnings, not errors. .
> 

Hi [per1234](https://github.com/per1234)
Thank you very much for your reply, much appreciated 
i got the warning in red color not white color so i thought it's error :)
good to know that i can go and try the code even with these warnings msgs.
please can you help me out , on Arduino H7 How can i set the servo output to 12-bit 333Hz? for 12bit output i used AnalogWriteResolution (12), and it look working but for setting the output at 333hz AnalogWriteFrequency (333) not working cuz i got errors when verifying the code, so what i need to use to get 333hz, the servo library standard output is 8bit with 50hz only, can i edit the servo.h file to get 333hz output ? but how?

please let me know!
Thanks

per1234, 2024-02-06:
>please can you help me out , on Arduino H7 How can i set the servo output to 12-bit 333Hz?

This is not an appropriate place to request assistance with your project. You are welcome to ask over on Arduino Forum. I'm sure we'll be able to help you out over there:

https://forum.arduino.cc/c/using-arduino/programming-questions/20

ghiathkamel, 2024-02-09:
to get 333Hz output i changed the servo.h file to this REFRESH_INTERVAL    3000

#define MIN_PULSE_WIDTH       544     // the shortest pulse sent to a servo  
#define MAX_PULSE_WIDTH      2400     // the longest pulse sent to a servo 
#define DEFAULT_PULSE_WIDTH  1500     // default pulse width when servo is attached
#define REFRESH_INTERVAL    3000     // minimum time to refresh servos in microseconds 
#define SERVOS_PER_TIMER       12     // the maximum number of servos controlled by one timer 
#define MAX_SERVOS   (_Nbr_16timers  * SERVOS_PER_TIMER)
#define INVALID_SERVO         255     // flag indicating an invalid servo index
```
after changing this, there is no warning messages anymore, i did not change it servo.cpp, i dont know if i need to but it works :)

razvanphp, 2024-09-11:
@ghiathkamel where exactly did you change it?

In my Arduino IDE 2.3.2 on Mac, if I control-click `Servo.h`, it opens as read-only from `/Users/razvan/Library/Arduino15/libraries/Servo/src/Servo.h`, so I don't think it uses this file. Did you recompile the Servo lib somehow manually?

Thanks!
R