arduino-libraries/Servo #147: Servo call from a class fails

ouwe-knutselaar, 2025-09-20:
I've an issue with controlling a servo from a class. I hope that I miss something that other people see ;-) 

I've the following code:

test.ino
```
#include "myservo.h"

Myservo testservo;
int teller;

void setup() {
  Serial.begin(9600);
};

void loop() {
  testservo.Swing();
}
```

myservo.h
```
#ifndef MYSERVO_H
#define MYSERVO_H

#include <Servo.h>

class Myservo{
  private:
    Servo servo;

  public:
   Myservo();
   void Swing();
};

#endif
```

myservo.cpp
```
#include <Arduino.h>
#include "myservo.h"

Myservo::Myservo(){
  servo.attach(9);
};

void Myservo::Swing(){
  int tel;
  Serial.println(servo.read());
  for(tel=0;tel<180;tel++){
    delay(15);
  }
};
```

Somehow, the servomotor doesn't do anything. When I run the debugger I also notice that the value of Myservo.servo is 255 after calling the constructor and 0 when I call the Myservo.Swing().
I looks as if the value of servo is reset to zero. 

So, I don't know what is failing here.

per1234, 2025-09-20:
Hi @ouwe-knutselaar. Thanks for your interest in this open source project. This issue tracker is only to be used to report bugs or feature requests specific to the project. This is not an appropriate place to request project support. The appropriate place for such requests is the Arduino Forum. I'm sure we will be able to help you out over there:

https://forum.arduino.cc/c/projects/programming/20

closed 2025-09-20 by per1234

ouwe-knutselaar, 2025-09-20:
I understand that, but I thought that it could be an issue in the library.

ouwe-knutselaar, 2025-09-21:
I see the error, I missed the ' servo.write(tel);'. But even then it did not work.
I posted the question on StackOverflow.